jeremiah-k/meshtastic-matrix-relay #42: Add info about node names and node details

vchrizz, 2023-05-12:
It would be nice to not only display the node count (displayed at first line output of `!health`) but also display the names/details of those nodes.
Maybe for better readability as an own listing in another command like `!nodes` ?
If possible, also display extended information. Some devices provide GPS coordinates, some devices like T-Echo provide also pressure. 
Thank you!

geoffwhittington, 2023-05-16:
Thanks for sharing your idea - can you provide a mock output example ?

vchrizz, 2023-05-16:
I am not sure what data can be optained or what data you are able to get from the device, whatever data can be shown could be interesting. Maybe something like that:
> Nodes: $nodecount
> $shortname $longname / $devicemodel / $battery $voltage / $rssi $snr / $lastseen [ / $optional telemetry data like gps, temperature, pressure, whatever the device provides]

for example:
input: 
> @botname: !nodes

bot output:
> Nodes: 5
> 2f8d CHiLtastic_2f8d / t-echo / 92% 4,07V / 6,0 / now / 48.24425 16.36195 / 24,6°C / 39% 991,6hPa
> 641c hei76tastic_641c / t-beam v1.1 / 98% 4,13V / -50 7,0 / 2 min
> a212 Solartastic_a212 / rak4631 / 73% 3,96V / -52 6,0 / 12 min
> b38c hei76rooftastic_b38c / t-beam v0.7 / 55% 3,82V / -77 6,5 / now / 48.24453 16.36211
> ec6d Meshtastic_ec6d / rak4631 / 98% 4,13V / -76 6,5 / 3 min

Thank you!

geoffwhittington, 2023-05-18:
Cool ! I'm thinking of pushing this up and iterating over it

[image]

geoffwhittington, 2023-05-18:
The challenge with this rendering is that mobile doesn't support tables. Is the ended use case for desktop or mobile? If the target use case is mobile, tables are not a good fit. Your example above is a better choice I think

geoffwhittington, 2023-05-18:
What do you think of

[image]

vchrizz, 2023-05-18:
I didn't even know that tables are possible. How does a table look like on mobile if tables are not supported?
I would see the target use case generic for every device. Depending if the table on mobile is somewhat readable, you could implement it as table and wait for the table-implementation on mobile. Or implement it as simple text and optimize as table as soon as table is supported on mobile.

Your screenshot looks nice, just that I would put every device in an own line.

geoffwhittington, 2023-05-18:
[image]

vchrizz, 2023-05-18:
👍 great! Maybe add some description, either add V to battery and dB to snr to know what the values mean :)

closed 2023-05-18 by geoffwhittington

geoffwhittington, 2023-05-18:
I've pushed up an initial implementation that is available in `0.3.22`

There are some missing data points like GPS location. I'm not jazzed about sharing that private detail in public locations but perhaps owners know best. I'm happy to discuss further

geoffwhittington, 2023-05-18:
Thank-you for your idea submission!

vchrizz, 2023-05-18:
Thank you for the update, I tried using the source from the repo, but for some reason it does not work. I wondered why it does not work and to be sure I downloaded the release you mentioned and installed it again from scratch, but still it does not work. When I send `@botname: !help` or `@botname: !nodes` in the matrix room the bot just sends this as typical message over to meshtastic.

vchrizz, 2023-05-19:
Ah, sorry, I didn't know you implemented that as plugins and expected the functionality to be enabled already. After I added them to the config, it works perfectly!
Thank you very much! Works really great!
Maybe you want to add the plugins to the sample_config.yaml ?

PR #44 (2023-05-18): Add nodes command
Fixes #42