jeremiah-k/meshtastic-matrix-relay #75: SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

fphammerle, 2024-08-15:
Hi,

thanks a lot for developing and maintaining meshtastic-matrix-relay!

On v0.7.0 with meshtasticd v2.3.13.83f5ba0 connected via TCP (both running on same raspberry pi) the following error keeps recurring: 
```python
Traceback (most recent call last):
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/util.py", line 288, in _run
    o()
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1297, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface
```

In my setup, all messages in Matrix and all manually typed message in Meshtastic are bridged.
However, messages sent by Meshtastic's detection sensor module are not bridged.
I do not know what this issue is related to the error above.

Any ideas? Thank you!

pip freeze:
```
aiofiles==23.2.1
aiohappyeyeballs==2.3.5
aiohttp==3.10.3
aiohttp-socks==0.7.1
aiosignal==1.3.1
appdirs==1.4.4
async-timeout==4.0.3
attrs==24.2.0
bleak==0.21.1
certifi==2024.7.4
charset-normalizer==3.3.2
contourpy==1.2.1
cycler==0.12.1
dbus-fast==2.22.1
dotmap==1.3.30
fonttools==4.53.1
frozenlist==1.4.1
future==0.18.3
geographiclib==2.0
h11==0.14.0
h2==4.1.0
haversine==2.8.0
hpack==4.0.0
hyperframe==6.0.1
idna==3.7
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
Logbook==1.7.0.post0
Markdown==3.4.3
matplotlib==3.9.0
matrix-nio==0.20.2
meshtastic==2.3.15
multidict==6.0.5
numpy==2.0.1
packaging==24.1
pexpect==4.9.0
Pillow==9.5.0
print-color==0.4.6
protobuf==5.27.3
ptyprocess==0.7.0
py-staticmaps==0.4.0
pycryptodome==3.20.0
pyparsing==3.1.2
Pypubsub==4.0.3
PyQRCode==1.2.1
pyserial==3.5
python-dateutil==2.9.0.post0
python-slugify==8.0.4
python-socks==2.5.0
PyYAML==6.0.2
referencing==0.35.1
requests==2.31.0
rpds-py==0.20.0
s2sphere==0.2.5
schedule==1.2.0
six==1.16.0
svgwrite==1.4.3
tabulate==0.9.0
text-unidecode==1.3
unpaddedbase64==2.1.0
urllib3==2.2.2
webencodings==0.5.1
yarl==1.9.4
```

RyanWor, 2024-09-24:
Also seeing this in the last month or so with 0.7.0 and meshtastic 2.3.14 using Heltec V3 over LAN/by IP. 

```
Traceback (most recent call last):
  File "/home/ryanw/meshtastic-matrix-relay/.pyenv/lib/python3.11/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ryanw/meshtastic-matrix-relay/.pyenv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ryanw/meshtastic-matrix-relay/.pyenv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ryanw/meshtastic-matrix-relay/.pyenv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/ryanw/meshtastic-matrix-relay/.pyenv/lib/python3.11/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface
```

pip freeze:
```
h2==4.1.0
haversine==2.8.0
hpack==4.0.0
hyperframe==6.0.1
idna==3.7
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
kiwisolver==1.4.5
Logbook==1.7.0.post0
Markdown==3.4.3
matplotlib==3.9.0
matrix-nio==0.20.2
meshtastic==2.3.14
multidict==6.0.5
numpy==2.0.1
packaging==24.1
pexpect==4.9.0
Pillow==9.5.0
print-color==0.4.6
protobuf==5.27.3
ptyprocess==0.7.0
py-staticmaps==0.4.0
pycryptodome==3.20.0
pyparsing==3.1.2
Pypubsub==4.0.3
PyQRCode==1.2.1
pyserial==3.5
python-dateutil==2.9.0.post0
python-slugify==8.0.4
python-socks==2.5.0
PyYAML==6.0.1
referencing==0.35.1
requests==2.31.0
rpds-py==0.19.1
s2sphere==0.2.5
schedule==1.2.0
six==1.16.0
svgwrite==1.4.3
tabulate==0.9.0
text-unidecode==1.3
typing_extensions==4.12.2
unpaddedbase64==2.1.0
urllib3==2.2.2
webencodings==0.5.1
yarl==1.9.4
```

jeremiah-k, 2024-10-01:
I apologize for the slow reply I haven't had much time to work on this project lately.  This had been an issue since the beginning and the error didn't affect functionality, so it was on the backburner.  I just dug into everything an refactored the connection handling.

> However, messages sent by Meshtastic's detection sensor module are not bridged.

I don't think this is related to that error.  I don't have any experience with the detection sensor module, myself. I'm guessing they're using something other than portnum TEXT_MESSAGE_APP.  That's the only packet type the relay looks for & uses for its core functionality. 

If you want, you can add this line inside the `on_meshtastic_message` function  (in meshtastic_utils.py) and capture some detection sensor packets to see how they're sent.
```
logger.info(f"Received packet from mesh: {packet}")
```

But as far as that error goes, when you have some time please test the changes in the `fixing-issue-75` branch and let me know how it works for you.  Thanks.

fphammerle, 2024-10-02:
> But as far as that error goes, when you have some time please test the changes in the fixing-issue-75 branch and let me know how it works for you.

I tested commit 6289fa2c797158094a50a838fe50e4acec13f031 on branch `fixing-issue-75`. `SenderUnknownMsgDataError` did not occur anymore. Bridging works in both directions. Thank you! 

After the first startup I once got:
```
ERROR:root:Error while parsing FromRadio bytes:b'…' Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1031, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'
ERROR:root:Error while handling message from radio Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 189, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1037, in _handleFromRadio
    raise ex
  File "/opt/meshtastic-matrix-relay/venv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1031, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'
```
However, I can't reproduce this error and bridging worked fine.

I will report back on the portnum of detection messages.

jeremiah-k, 2024-10-03:
I worked on it a bit more and the most recent changes were merged into `main`

fphammerle, 2024-10-04:
Thank you!

I created a PR for detection messages.

closed 2024-10-06 by jeremiah-k