jeremiah-k/meshtastic-matrix-relay #122: Exception in _writeBytes

troed, 2025-02-23:
(This seems very similar to the closed issue #62)

I'm seeing the following exception happen every so often, and I then have to manually restart the bridge to get the messages that came in during this time.

Hardware: Heltec v3
Firmware: 2.5.20.4c97351 (latest as of posting this)

Connection between bridge and meshtastic node is wifi, using meshtastic.local.

`
Exception in thread Thread-13:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1431, in run
    self.function(*self.args, **self.kwargs)
  File "/home/matrix/meshtastic-matrix-relay/.pyenv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1116, in callback
    self.sendHeartbeat()
  File "/home/matrix/meshtastic-matrix-relay/.pyenv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1105, in sendHeartbeat
    self._sendToRadio(p)
  File "/home/matrix/meshtastic-matrix-relay/.pyenv/lib/python3.12/site-packages/meshtastic/mesh_interface.py", line 1180, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/matrix/meshtastic-matrix-relay/.pyenv/lib/python3.12/site-packages/meshtastic/stream_interface.py", line 120, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/matrix/meshtastic-matrix-relay/.pyenv/lib/python3.12/site-packages/meshtastic/tcp_interface.py", line 79, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
`

wioxjk, 2025-02-23:
Facing the exact same issue with a Heltec WSL v3

takov751, 2025-02-27:
Had the same issue. I noticed that it was probably my android meshtastic app connecting aggressively to the meshtastic device over TCP. I suspect the problem is in meshtastic_utils.py threading handling. I will try to check it out later

jeremiah-k, 2025-03-01:
> (This seems very similar to the closed issue [#62](https://github.com/geoffwhittington/meshtastic-matrix-relay/issues/62))
> 

Yes, this has been an issue for awhile. I should've reopened the other issue when I realized it was still a problem.  I can get it to detect disconnection/reconnect reliably with serial, but I can't get it to detect disconnection reliably in order to reconnect using network (tcp) interface. 

The meshtastic library doesn't raise an exception or try to repair the connection.  Other projects have written a heartbeat as a workaround, but I haven't ever been able to get it working properly.  I see they've been talking about the issue in #python in the Meshtastic Discord recently.

I haven't had much free time to work on projects lately since work picked up again recently. If anyone wants to take a look and see what can be improved upon for tcp connection handling (here and upstream maybe?) , it would be much appreciated.

troed, 2025-03-02:
Thanks @jeremiah-k - with that explanation I understand the issue. Since it's TCP it's really the upstream library (whomever owns the socket) that should handle this. I will take a look.

mr-suw, 2025-03-08:
meshing-around (https://github.com/SpudGunMan/meshing-around) is doing a TCP reconnection after 15 seconds. I think it uses a watchdog mechanism by periodically requesting the node firmware and when that fails it reconnects. For reference the watchdog function is in system.py at line 1191 (https://github.com/SpudGunMan/meshing-around/blob/main/modules/system.py)

jeremiah-k, 2025-03-08:
> meshing-around (https://github.com/SpudGunMan/meshing-around) is doing a TCP reconnection after 15 seconds. I think it uses a watchdog mechanism by periodically requesting the node firmware and when that fails it reconnects.

Thanks. I had talked to SpudGunMan and mian in the Discord awhile back and tried implementing something like his [getNodeFirmware](https://github.com/SpudGunMan/meshing-around/blob/c160678e7921deff88bdffe50bd9802428cdd0e1/modules/system.py#L847-L859) function, but couldn't get it to detect tcp disconnection properly at the time.

I should be able to make some time in the next few days to take a closer look at his implementation and try again.

jeremiah-k, 2025-03-08:
I had some time today and I think I might've finally got it.  For anyone who can test, please fetch and checkout `tcp-reconnection-2` branch and see if reconnects properly for you.

jeremiah-k, 2025-03-09:
> I had some time today and I think I might've finally got it. For anyone who can test, please fetch and checkout `tcp-reconnection-2` branch and see if reconnects properly for you.

Before I wasn't letting it run and was just testing with remote admin command reboots, but after testing `tcp-reconnection-2` branch some more, there's a timing issue in task scheduling causing unnecessary reconnections. There may also be a concurrency or threading issue affecting how reconnections are handled.

At least I’m on the right track now. Hopefully I can get this resolved soon.

troed, 2025-03-12:
Looks good to me, I can see reconnection taking place successfully in the log output! Many thanks :) 

```
2025-03-12 02:58:27 +0100 ERROR:Meshtastic:Network connection lost: Timed out waiting for an acknowledgment
2025-03-12 02:58:27 +0100 ERROR:Meshtastic:Lost connection. Reconnecting...
2025-03-12 02:58:27 +0100 INFO:Meshtastic:Reconnection attempt starting in 10 seconds...
2025-03-12 02:58:27 +0100 INFO:Meshtastic:Reconnection already in progress. Skipping additional reconnection attempt.
2025-03-12 02:58:37 +0100 INFO:Meshtastic:Connecting to host meshtastic.local ...
2025-03-12 02:58:39 +0100 INFO:Meshtastic:Connected to TS1 / HELTEC_V3
2025-03-12 02:58:39 +0100 INFO:Meshtastic:Reconnected successfully.
2025-03-12 06:43:09 +0100 ERROR:Meshtastic:Network connection lost: Timed out waiting for an acknowledgment
2025-03-12 06:43:09 +0100 ERROR:Meshtastic:Lost connection. Reconnecting...
2025-03-12 06:43:09 +0100 INFO:Meshtastic:Reconnection attempt starting in 10 seconds...
2025-03-12 06:43:09 +0100 INFO:Meshtastic:Reconnection already in progress. Skipping additional reconnection attempt.
2025-03-12 06:43:19 +0100 INFO:Meshtastic:Connecting to host meshtastic.local ...
2025-03-12 06:43:22 +0100 INFO:Meshtastic:Connected to TS1 / HELTEC_V3
2025-03-12 06:43:22 +0100 INFO:Meshtastic:Reconnected successfully.
2025-03-12 07:13:10 +0100 ERROR:Meshtastic:Network connection lost: Timed out waiting for an acknowledgment
2025-03-12 07:13:10 +0100 ERROR:Meshtastic:Lost connection. Reconnecting...
2025-03-12 07:13:11 +0100 INFO:Meshtastic:Reconnection attempt starting in 10 seconds...
2025-03-12 07:13:11 +0100 INFO:Meshtastic:Reconnection already in progress. Skipping additional reconnection attempt.
2025-03-12 07:13:21 +0100 INFO:Meshtastic:Connecting to host meshtastic.local ...
2025-03-12 07:13:23 +0100 INFO:Meshtastic:Connected to TS1 / HELTEC_V3
2025-03-12 07:13:23 +0100 INFO:Meshtastic:Reconnected successfully.
2025-03-12 08:51:59 +0100 INFO:Meshtastic:Processing inbound radio message from !ea067f88 on channel 0
```

(Since others have commented as well I'm not closing the issue right now. Your choice if you want to wait for more input or if you feel this is enough and close it yourself)

jeremiah-k, 2025-03-12:
Great to hear it! I'm able to stay connected to my rooftop solar node via BLE now, as well. I've had a decent amount of testing & positive feedback on Matrix too.

If anyone else here has any problems let me know, but I think I can close this for now.

Closed by #133

closed 2025-03-12 by jeremiah-k