jeremiah-k/meshtastic-matrix-relay #146: protobuf parsing error prevents startup

kruthoff, 2025-04-10:
Hi meshtastic-matrix-relay

The relay was running for weeks without issues, great! However, it doesn't connect to the meshtastic device via BLE anymore, and reports this protobuf error. I've rebooted the device and have tried to connect, but no luck.

Thanks for your support!

[mmrelay-error.txt](https://github.com/user-attachments/files/19679895/mmrelay-error.txt)

jeremiah-k, 2025-04-10:
> The relay was running for weeks without issues, great! However, it doesn't connect to the meshtastic device via BLE anymore, and reports this protobuf error. I've rebooted the device and have tried to connect, but no luck.

It's good to hear that that the relay's been running well overall!

The protobufs are handled by the `meshtastic` python package, so running the commands below inside your virtualenv should take care of the problem.
```
pip install --upgrade meshtastic
```
and then this one to make sure the rest of your installed dependencies match the pinned versions:
```
pip install -r requirements.txt
```

I'm also in the process of packaging everything properly and hope to have a release today or tomorrow at the latest, when that happens you'll be able to install with pip/pipx and have all of this handled automatically.

kruthoff, 2025-04-11:
Hey! Thanks for the fast reply. You were right, my dependencies were not in a proper state.
Works as expected now.

closed 2025-04-11 by kruthoff