jeremiah-k/meshtastic-matrix-relay #180: High CPU usage (sometimes)

Finn10111, 2025-05-17:
After some hours, I don't know exactly what could be the trigger, I see a high CPU usage caused by mmrelay. 

[image]

I am running it in docker because it was so convenient some time ago, with 1.0 it probably should not be necessary anymore. These are the only logs I can get:

```
meshtastic-relay-finn  | 2025-05-17 19:54:40 +0000 INFO:db_utils:Creating new database at: /root/.mmrelay/data/meshtastic.sqlite
meshtastic-relay-finn  | 2025-05-17 19:54:40 +0000 INFO:Plugins:Checking plugin config...
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 DEBUG:Plugin:health:Started with priority=10
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 DEBUG:Plugin:ping:Started with priority=10
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 DEBUG:Plugin:telemetry:Started with priority=10
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 DEBUG:Plugin:weather:Started with priority=10
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 DEBUG:Plugin:nodes:Started with priority=10
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Plugins:Plugins loaded: health, ping, telemetry, weather, nodes
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 WARNING:Meshtastic:Using 'network' connection type (legacy). 'tcp' is now the preferred name and 'network' will be deprecated in a future version.
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Meshtastic:Connecting to host 192.168.1.195 ...
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Meshtastic:Connected to FINN / SEEED_XIAO_S3
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Matrix:Joined room '#meshtastic-finn:f2n.me' successfully
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Matrix:Listening for inbound Matrix messages...
meshtastic-relay-finn  | 2025-05-17 19:54:41 +0000 INFO:Matrix:Starting Matrix sync loop...
```

jeremiah-k, 2025-05-17:
If you're running a `mmrelay` release with a `meshtastic` python library version older than 2.6.2, high CPU usage was reported when connecting via TCP https://github.com/meshtastic/python/pull/761

The latest mmrelay release (1.0.7) includes the latest `meshtastic` release (2.6.3), which has been fixed.

It's easy to switch over, just follow the quick start instructions:
```
# Install using pipx for isolated installation (recommended)
pipx install mmrelay

# Generate a sample configuration file & then edit it
mmrelay --generate-config

# Start the relay (without --install-service to run manually)
mmrelay --install-service
```

Except after running `mmrelay --generate-config` copy your current config.yaml to ~/.mmrelay/config.yaml 

Then run `mmrelay --install-service`, it will set up a systemd user service and lists the commands to manage it easily. After that it's just `pipx upgrade mmrelay` to upgrade in the future.

Otherwise, if you can shell into your docker container and run `pip install --upgrade meshtastic` it should take care of it too.

Finn10111, 2025-06-17:
Oh yes, sorry for the missing feedback. I can confirm the CPU usage went back to normal after upgrading.